Ameba-AIoT/ameba-arduino-d #263: UDP.parsePacket() is returning wrong values

mardecker, 2025-02-20:
### Boards

BW16

### External Hardware

Ai-Thinker BW16 (RTL8720DN) 

### Hardware Configuration

just connected to WPA2/WPA3 Wifi

### Version

latest master (checkout manually)

### IDE Name

Arduino 1.8.19

### Operating System

Ubuntu

### Auto Flash Mode

Disable

### Erase All Flash Memory (4MB)

Disable

### Standard Lib

Arduino_STD_PRINTF

### Upload Speed

1500000

### Description

I was trying to use a CoAP Library when I noticed, that no CoAP packages were recieved correctly. It seems like, that the funktion udp.parsePacket() just returns a solid 1, when Data is recieved.

### Sketch

```cpp
#include <Arduino.h>
#include <WiFi.h>
#include <WiFiUdp.h>
#include <coap-simple.h>

#define MAX_SEND_SIZE 256
#define MAX_RECV_SIZE 1500
#define UDP_SERVER_PORT 5683

char ssid[] = "wland";       // dein WLAN-SSID
char password[] = "Asylantrag2020";  // dein WLAN-Passwort

IPAddress coapServer(192, 168, 178, 82);  // CoAP-Server-IP
int coapPort = 5683;  // CoAP-Standard-Port

WiFiUDP udp;  // UDP-Objekt für CoAP
Coap coap(udp, MAX_SEND_SIZE);  // CoAP-Client-Instanz

// **Callback-Funktion vorher deklarieren**
void callback(CoapPacket &packet, IPAddress ip, int port);

void setup() {
    Serial.begin(115200);
    
    // WLAN-Verbindung herstellen
    Serial.print("Verbinde mit WLAN...");
    WiFi.begin(ssid, password);
    
    while (WiFi.status() != WL_CONNECTED) {
        delay(500);
        Serial.print(".");
    }
    Serial.println(" Verbunden!");

    // CoAP-Client starten
    coap.start();

    // Antwort-Callback setzen
    coap.response(callback);

    
    // CoAP GET Anfrage senden
    Serial.println("Sende CoAP GET-Anfrage...");
    coap.get(coapServer, coapPort, "temp/1");
}

void loop() {
    int packetSize = udp.parsePacket();
    if (packetSize) {
        Serial.print("UDP-Paket empfangen: ");
        Serial.println(packetSize);
        
        char packetBuffer[512];  // Puffer für eingehende Daten
        int len = udp.read(packetBuffer, 512);
        if (len > 0) {
            packetBuffer[len] = '\0';  // String terminieren
        }

        Serial.print("Daten: ");
        Serial.println(packetBuffer);
    }

    coap.loop();  // CoAP-Verarbeitung
}
// **Callback für die Antwort des CoAP-Servers**
void callback(CoapPacket &packet, IPAddress ip, int port) {
    Serial.print("Antwort erhalten von: ");
    Serial.println(ip);
    
    Serial.print("Port: ");
    Serial.println(port);

    Serial.print("Daten: ");
    for (int i = 0; i < packet.payloadlen; i++) {
        Serial.print((char)packet.payload[i]);
    }
    Serial.println();
}
```

### Error/Debug Message

```plain
I got no explicit message.
I just debugged the data via Serial.println (see provided Ino)
```

### Reproduce remarks

if you add CoAP Simple Library, you have to change ' #include "Udp.h" ' to ' #include "WiFiUdp.h" '. 

### I have checked online documentation, FAQ, GitHub Wiki and existing/closed issues.

- [x] I confirm I have checked online documentation, FAQ, GitHub Wiki and existing/closed issues.

mardecker, 2025-02-21:
Same Problem with the WiFiUdpSendReceiveString example

fastfourier666, 2025-02-22:
I am finding this too, although `udp.read` does return the correct length.