Ameba-AIoT/ameba-arduino-d #266: I want to hire someone who can port a Library to RTL8720dn

A3ST1CODE, 2025-02-28:
### Hardware Environment

RTL8720DN BW16

### Related technical area

DNS

### Is the feature request related to an un-fixed issue?

I am currently making a project for RTL8720dn that requires Captive Portal. On Espressif MCU, it can be made by using DNSServer library, but AFAIK there is no official library for RTL8720dn.

### Describe the ideal solution/reault/output.

-

### I have checked existing/closed list of issues and the Contributing Guide.

- [x] I confirm I have checked existing/closed list of issues and the Contributing Guide.

rhkesuma, 2025-03-20:
If there aren’t any working methods or libraries for the RTL8720DN/BW16, why not just create an easier solution, bruv? No need to do extensive work like making a complete, fully functional DNS server—just write some dnsHijacking code. That should help redirect any client connected to the AP straight to the captive portal you’ve already set up. I mean, if all you need is a fully working DNS server just for redirection, why overcomplicate it?

A3ST1CODE, 2025-03-21:
I don't have one and googling didn't give me anything useful. Can you provide a code example?

rhkesuma, 2025-03-22:
You probably won’t find any parts of it on Google. That’s why I said in an earlier post—why don’t you just try coding/creating it yourself?

This small BIN file, [km0_km4_image2.bin](https://github.com/rhkesuma/dnsJack/blob/main/km0_km4_image2.bin), is proof of a working code/concept for custom, tailor-made DNS hijacking. Just flash it and see if it works as you expect, since the BW16 never had non-NDA support for DNSServer.h.

Warm regards from Kios Martabak Bangka Lumenta, Bandar Lampung.

A3ST1CODE, 2025-03-22:
thanks, can you provide the source?

Best regards

rhkesuma, 2025-03-22:
No, it’s definitely not a flash dump or taken from someone else’s firmware. If it were as easy as extracting and reusing code from a dumped flash without the necessary libraries for compilation, there would already be plenty of dnsServer libraries available for the RTL8720. At the very least, you could have ripped it from a dump, loaded it into Ghidra, and made it work yourself. I’m pretty sure you could do it without much hassle, since it’s so easy peasy.

What I have is a compiled version of a working Simple DNS Hijack code, proving that I can redirect UDP packets by intercepting and modifying DNS responses for specific domains. This is done by capturing UDP packets on port 53 (DNS queries) and responding with a custom IP address that points to the captive portal using the following method:

```
FakeDNSresponse[] = {  
    0xC0, 0x0C, 0x00, 0x01, 0x00, 0x01, 0x00, 0x00, 0x00, 0x3C, 0x00, 0x04, 192, 168, 1, 1 
};
void sendFakeDNS(struct udp_pcb *pcb, struct pbuf *p, const ip_addr_t *addr, u16_t port) {
    if (!p || !addr) return;

    struct pbuf *response = pbuf_alloc(PBUF_TRANSPORT, p->len + sizeof(fakeDNSResponse), PBUF_RAM);
    if (!response) return;

    memcpy(response->payload, p->payload, p->len);
    uint8_t *res = (uint8_t *)response->payload;
    res[2] |= 0x80; 
    res[3] |= 0x80;
    res[7] = 1; 
```
fakeDNSresponse[]. Simulates a DNS response to redirect clients to a specific IP address (192.168.1.1, BW16 default IP).

Then create response packet:
A new p_buf/packet buffer (response) is allocated for the DNS response. The response size is determined by the length of fakeDNSresponse. also with the udp_pcb/protocol control block for UDP, and so on, so on... 

As for why I compiled the code with an open AP using your name as the SSID, with no credentials—it was simply to prove a point. The captive portal page itself can be anything but empty, so I designed it with a form layout that you might be 'familiar' with.

Before assuming that I stole someone’s work by reverse-engineering a dumped firmware, please be patient. I’ll upload the Simple DNS Hijacking code along with an example when it’s ready. The code is still undergoing multiple fixes, patches, and refinements, as it is still in early development.

plus, I’m not a programmer or software engineer, and my progress has been slowed down by my primary job as a Martabak Bangka seller.

p.s. sorry for my bad english, as english is not my first language.

A3ST1CODE, 2025-03-22:
That's okay, I understand you, I'm very grateful for your understanding of that. can you share your contact.