Ameba-AIoT/ameba-arduino-d #276: After updating the TensorFlow patch, an error occurred when running the Amazon_AWS_IoT_Basic example.

cold63, 2025-03-29:
### Boards

Datalogger (RTL8720DF)

### External Hardware

none

### Hardware Configuration

none

### Version

latest dev (checkout manually)

### IDE Name

Arduino IDE 2.3.4

### Operating System

Windows 10

### Auto Flash Mode

Enable

### Erase All Flash Memory (4MB)

Disable

### Standard Lib

Arduino_STD_PRINTF

### Upload Speed

1500000

### Description

Using the latest dev version 3.1.8 build20250321, after updating the TensorFlow patch, an error occurred when running the Amazon_AWS_IoT_Basic example.

### Sketch

```cpp
/*

 Basic Amazon AWS IoT example

 Example guide:
 https://www.amebaiot.com/en/amebad-arduino-aws-shadow/
 */

#include <WiFi.h>
#include <PubSubClient.h>

// Update these with values suitable for your network.
char ssid[] = "Network_SSID";       // your network SSID (name)
char pass[] = "Password";           // your network password
int status = WL_IDLE_STATUS;        // Indicator of Wifi status

WiFiSSLClient wifiClient;
PubSubClient client(wifiClient);

#define THING_NAME "amebaDevBoard"

char mqttServer[]     = "a4adrtgyzxov42-ats.iot.ap-southeast-1.amazonaws.com";
char clientId[]       = "amebaClient";
[... 245 lines cut ...]
     return publish_P(topic, (const uint8_t*)payload, payload ? strnlen(payload, this->bufferSize) : 0, retained);
                                                                                                 ^
C:\Users\Dell\AppData\Local\Arduino15\packages\realtek\hardware\AmebaD\3.1.8-build20250321\libraries\MQTTClient\src\PubSubClient.cpp: In member function 'boolean PubSubClient::publish_P(const char*, const uint8_t*, unsigned int, boolean)':
C:\Users\Dell\AppData\Local\Arduino15\packages\realtek\hardware\AmebaD\3.1.8-build20250321\libraries\MQTTClient\src\PubSubClient.cpp:662:43: error: 'strnlen' was not declared in this scope
     tlen = strnlen(topic, this->bufferSize);
                                           ^
C:\Users\Dell\AppData\Local\Arduino15\packages\realtek\hardware\AmebaD\3.1.8-build20250321\libraries\MQTTClient\src\PubSubClient.cpp: In member function 'boolean PubSubClient::subscribe(const char*, uint8_t)':
C:\Users\Dell\AppData\Local\Arduino15\packages\realtek\hardware\AmebaD\3.1.8-build20250321\libraries\MQTTClient\src\PubSubClient.cpp:779:57: error: 'strnlen' was not declared in this scope
     size_t topicLength = strnlen(topic, this->bufferSize);
                                                         ^
C:\Users\Dell\AppData\Local\Arduino15\packages\realtek\hardware\AmebaD\3.1.8-build20250321\libraries\MQTTClient\src\PubSubClient.cpp: In member function 'boolean PubSubClient::unsubscribe(const char*)':
C:\Users\Dell\AppData\Local\Arduino15\packages\realtek\hardware\AmebaD\3.1.8-build20250321\libraries\MQTTClient\src\PubSubClient.cpp:828:54: error: 'strnlen' was not declared in this scope
  size_t topicLength = strnlen(topic, this->bufferSize);
                                                      ^

使用檔案夾: C:\Users\Dell\AppData\Local\Arduino15\packages\realtek\hardware\AmebaD\3.1.8-build20250321\libraries\WiFi  內的程式庫 WiFi 版本 1.0.0 
使用檔案夾: C:\Users\Dell\AppData\Local\Arduino15\packages\realtek\hardware\AmebaD\3.1.8-build20250321\libraries\MQTTClient  內的程式庫 AmebaMQTTClient 版本 1.0.0 
exit status 1

Compilation error: exit status 1
```

### I have checked online documentation, FAQ, GitHub Wiki and existing/closed issues.

- [x] I confirm I have checked online documentation, FAQ, GitHub Wiki and existing/closed issues.

DeqingSun, 2025-05-07:
I got a similar issue. 

3.1.8 has error like:

```
n file included from /Users/deqinguser/Library/Arduino15/packages/realtek/hardware/AmebaD/3.1.8/cores/ambd/Arduino.h:46:0,
                 from /private/var/folders/ph/rd4h23bd74b37mz4j8ddl6l40000gp/T/arduino/sketches/7CBDD1113DD9316C881219CACF0FE491/sketch/BW16_TEST_20250429_AWS_TEST.ino.cpp:1:
/Users/deqinguser/Library/Arduino15/packages/realtek/tools/ameba_d_asdk_toolchain/1.0.1/arm-none-eabi/include/c++/6.5.0/bits/stl_vector.h: In member function 'std::vector<_Tp, _Alloc>::size_type std::vector<_Tp, _Alloc>::_M_check_len(std::vector<_Tp, _Alloc>::size_type, const char*) const':
/Users/deqinguser/Library/Arduino15/packages/realtek/hardware/AmebaD/3.1.8/cores/ambd/wiring_constants.h:78:19: error: expected unqualified-id before '(' token
 #define max(a, b) ((a) > (b) ? (a) : (b))
```

M-ichae-l, 2025-06-12:
TensorFlow patch is  for TensorFlow realted example, it may not compatible with Amazon_AWS_IoT_Basic example